CapnBry/CRServoF #1: add USB HID Joystick Support

pitts-mo, 2021-05-22:
Is there interest in adding a feature similar to the HID Joystick Support found in betaflight 3.4+?

I am not expecting there will be enough desire to yield implementation. 

Thank you for building this @CapnBry,
-p

CapnBry, 2021-05-22:
I had interest in it, but there is no HID Joystick support built into the stmduino core and I wasn't about to add it for something that I didn't think anyone would use (this project). If someone is looking for just RX to HID Joystick, there are other options already. Like PeHo's [RC Receiver to USB HID](https://github.com/cruwaller/rc_receiver_to_usb_hid), which uses the same connections so just flash and go.

pitts-mo, 2021-05-23:
Thank you @CapnBry for pointing me to PeHo's [https://github.com/cruwaller/rc_receiver_to_usb_hid](url). I had desired something similar for quite a while and I recently setup a half dead flight controller w/betaflight to test ELRS on simulators... As for your CRServoF project I plan to take some time to see if can get it running on an old CC3D for testing ERLS on a basic foam board wing :-). 

Do you think this issue should be closed or maybe let it sit a while for others to reference? 
Thank you,
-p

CapnBry, 2021-05-23:
We can leave it open, and maybe if there's a lot of people using CRServoF then I'll pull the code in from Jye to add the joystick support as well. In ELRS 1.1 we'll almost certainly have PWM output support so I expect this project will be short-lived.

sfjuocekr, 2022-01-16:
If you have a board that supports USB, I used a Teensy 3.2, you could try the HID joystick I made using this CrsfSerial library.

It might work for other boards as well with some minor modifications.

benedikt-bartscher, 2023-03-10:
see https://github.com/mikeneiderhauser/CRSFJoystick

kkbin505, 2024-12-04:
I use a arduino pro micro to implement crsf to HID, the only modification needed is to flash the RX with 115200 baud rate, because the arduino do not support 4k baudrate.  It works well with the simulator like tinywhoop go.
https://github.com/kkbin505/Simple_RX

CapnBry, 2024-12-04:
> I use a arduino pro micro to implement crsf to HID, the only modification needed is to flash the RX with 115200 baud rate, because the arduino do not support 4k baudrate. It works well with the simulator like tinywhoop go. https://github.com/kkbin505/Simple_RX

Nice! Atmega328 support higher baud rates as well though, 256000 and 512000. I think both of these work on the CrsfSerial side (the first for sure, the second I can't remember if it did) and can lower the joystick latency by a couple milliseconds. Also requires setting the RX to matching baud like you have done, and that can be done in the RX's webui as well.