CapnBry/CRServoF #8: Should we design a custom PCB for this?

crteensy, 2021-08-08:
The bluepill is rather large and certainly not tailored to this application.

Would you like to work on a custom PCB? I could sketch something in KiCad, maybe with a smaller MCU, and make it better and more compact. I already like the fact that there's an analog input for the battery voltage, and we could pave the way for current measurements, too.

CapnBry, 2021-08-08:
The reason I didn't bother is: where you gonna get the STM32F103 to populate it with? :-D

I know there is a commercial manufacturer that is working on a design as well, but all I've heard from them is them asking permission to do it.

crteensy, 2021-08-08:
for DIY, the chip could always come from a bluepill :D 
But you're right, they aren't available at the moment. But rather have a design ready when the chip can be bought than having to rush a design later.

derFrickler, 2021-09-10:
PCB would be great! Maybe even with a BEC integrated or a header to solder one of these very small buck converter PCBs.
I ordered some of these boards from Ali to try your code:
https://stm32-base.org/boards/STM32F103C8T6-Purple-Pill.html
The proto-holes should make it easy to solder a 3-row pinheader for the servo connection.

crteensy, 2021-09-10:
I had posted one in the ELRS discord, with BEC and a 1-3S 15A ESC included. Lots of possibilities here - but just an MCU with servo output and BEC is certainly possible too.

CapnBry, 2021-12-14:
I'm going to close this because we have commercial CRSF to PWM converters now and PWMP direct PWM RX so I don't think we need any other designs going in this direction.

closed 2021-12-14 by CapnBry