CapnBry/CRServoF #11: Target for CJMCU1038 - Purple Pill

jhermen-List, 2021-09-22:
I have some of these Purple Pill boards lying around:
https://stm32-base.org/boards/STM32F103C8T6-Purple-Pill.html
My Idea was to use them for your project, as they are small and have a proto area which makes it easy to solder a 6x3 Pin Header for the Servos. 

I tried to make a new Target for the board using the pins that are broken out on the board. 
``` 
#elif defined(TARGET_CJMCU1038)
      #define DPIN_LED        PB_11
      #define LED_INVERTED    1
      #define APIN_VBAT       PB_0
      #define USART_INPUT     USART1  // UART1 RX=PA10 TX=PA9
      #define OUTPUT_PIN_MAP  PA_0, PA_1, PA_2, PA_3, PB_3, PB_2, PB_1, PA_4 
 #endif
```
I am able to compile and flash the code, it seems to connect to the receiver but i don't get the right servo outputs.
I enabled some printlines in your code and the serial console says: 
```
ADC=0 0V
ADC=0 0V
6666
�␌␔␀␀␀␀␀␄␀␀␀␀��␌␔␀␀␀␀␀␄␀␀␀␀ADC=0 0V
ADC=0 0V
ADC=0 0V
ADC=0 0V
tentative conn
30dBm
```
And then it seems stuck.

On pins PA_0, PA_1, PA_2, PA_3 i can see a 50Hz servo signal with 1700ms. The other pins show nothing. 
I guess its some problem due to the used timers for my pins that need to be changed. Maybe you can point me in the right direction.

Thanks a lot 
       Johannes

jhermen-List, 2021-09-22:
please delete, double post.

closed 2021-09-22 by jhermen-List

CapnBry, 2021-09-22:
The pins you use aren't just randomly selected. You have to match them to a hardware timer because they are not assignable. Did you figure this out? Because you've closed the issue. I can reopen it if github won't let you.

derFrickler, 2021-09-22:
This ticked was done on my wrong account - sorry, please delete it if possible