CapnBry/CRServoF #12: Target for CJMCU1038 - Purple Pill

derFrickler, 2021-09-22:
I have some of these Purple Pill boards lying around:
https://stm32-base.org/boards/STM32F103C8T6-Purple-Pill.html
My Idea was to use them for your project, as they are small and have a proto area which makes it easy to solder a 6x3 Pin Header for the Servos. 

I tried to make a new Target for the board using the pins that are broken out on the board. 
``` 
#elif defined(TARGET_CJMCU1038)
      #define DPIN_LED        PB_11
      #define LED_INVERTED    1
      #define APIN_VBAT       PB_0
      #define USART_INPUT     USART1  // UART1 RX=PA10 TX=PA9
      #define OUTPUT_PIN_MAP  PA_0, PA_1, PA_2, PA_3, PB_3, PB_2, PB_1, PA_4 
 #endif
```
I am able to compile and flash the code, it seems to connect to the receiver but i don't get the right servo outputs.
I enabled some printlines in your code and the serial console says: 
```
ADC=0 0V
ADC=0 0V
6666
�␌␔␀␀␀␀␀␄␀␀␀␀��␌␔␀␀␀␀␀␄␀␀␀␀ADC=0 0V
ADC=0 0V
ADC=0 0V
ADC=0 0V
tentative conn
30dBm
```
And then it seems stuck.

On pins PA_0, PA_1, PA_2, PA_3 i can see a 50Hz servo signal with 1700ms. The other pins show nothing. 
I guess its some problem due to the used timers for my pins that need to be changed. Maybe you can point me in the right direction.

Thanks a lot 
       Johannes

CapnBry, 2021-09-22:
The pins you use aren't just randomly selected. You have to match them to a hardware timer because they are not assignable. Did you figure this out? Because you've closed the issue. I can reopen it if github won't let you.

derFrickler, 2021-09-22:
Thanks for the fast reply!
I just chose the pins that were broken out on the board where i needed them, did not care about the timers. 
Its my first STM32 project, i only did AVR and Arduino before where that has never been an issue for me. 
Can you give me a hint on where to find that info and how to change the timers?

Thanks a lot
      Johannes

derFrickler, 2021-09-23:
Thanks, 
Found the info and managed to get that done, had 2 pins in use that did not support hardware PWM.

Now i get the correct receiver values and have a PWM on all pins - but still only 1700ms independent of the receiver values. 
servoSetUs seems to get the wrong µs values, but the serial debugging in packetChannels show the right values . 
Must be somewhere around where it messes up.
https://github.com/CapnBry/CRServoF/blob/caddba34d603e75e339e7f37d813cf1ccaafcb49/src/main.cpp#L73
I will check that out this evening.

CapnBry, 2021-09-23:
Good catch! I apparently put this bug in when I added the inverted output a couple weeks ago. Tested the inverted case, did not test that I left an assignment off the regular case. I just pushed the change to fix it, thanks!

derFrickler, 2021-09-23:
OK, that explains a lot, was thinking how that line of code was supposed to work....

derFrickler, 2021-09-23:
So, my prototype is running. I will add the target for that and a wiring diagram if you want. 
[image]
Will try to add an option to show the cell-voltage (divide by recognized cell number) instead of the full battery voltage in the telemetry as well.

CapnBry, 2021-09-23:
Nice! Sure if you want to do a PR for the target and include a markdown table for the pinout like there is a in the readme for the "F103" target. I guess maybe call this target purplepill with a comment that it is `CJMCU-1038`?

As far as cell voltage for VBatt, sure do this using a `constexpr bool report_cell_voltage`  and a `#define VBAT_MAX_CELL_VOLTAGE 430` lower down for being able to adjust the cell calculation. Keep all the math in integer centivolts.

JupaCreations, 2021-09-24:
derFrickler are the SLC and SDA pins available to connect for example the BMP280 or better baro I2C board?
Would that be possible to have a diy 8 channel LOS receiver with voltage and height measurement?

I have a blue pill to try but will also order  this purple pill... :-)

CapnBry, 2021-09-24:
> derFrickler are the SLC and SDA pins available to connect for example the BMP280 or better baro I2C board?
> Would that be possible to have a diy 8 channel LOS receiver with voltage and height measurement?

I would be against turning this into some sort of flight controller by adding extra sensors, so do not add this to the PR for adding this target.

Paco: The hardware is exactly the same between the bluepill and this. This only has the place to solder servo pinheader on (but is missing some of the pins the bluepill version uses).

derFrickler, 2021-09-25:
OK, i did a pull request for the Purplepill target. https://github.com/CapnBry/CRServoF/pull/13

I added a wiring pic to my repo aswell, if you want i can do a pull request for that aswell:
https://github.com/derFrickler/CRServoF/blob/master/doc/Purplepill_CJMCU1038_wiring.jpg

Next thing i will do is the cellcount detection and an report_cell_voltage option. Will do that in a separate request later. 

Baro and other sensors is definitely not on my list, especialy for a Vario its muchmore than just reading sensor values...

P.S. Seems that Matec will bring a board Servo PWM board soon: https://fpv-racing-forum.de/thread/13823-expresslrs/?postID=319659#post319659