CapnBry/CRServoF #14: Servos not working

JupaCreations, 2021-10-01:
Bluepill STM32F103C8 is connected to a HM EP1 RX. 
TX to RX and vice versa UART connection.
EP1 communicates by CRSF with the 2.4 TX as I get nice 1RRSi values on the TX.
By ST-Link V2 I uploaded the firmware.bin file created from the PIO F103_serial folder by Platformio.
For uploading I used the ST32CubeProgrammer.
Program says upload succeed.
Yet I get no servo movement.
During programming the yellow BOOT pins I did not change as it is not mentioned although many YT videos show this with UART connection. But with the ST-Link it seems not needed, but I could be totally wrong. I also tried it with BOOT1 in HIGH mode, but same thing.

What more can I do?

CapnBry, 2021-10-01:
I just flashed master and servo outputs are working for me. Make sure you aren't flashing the version from last week maybe as that had a bug that was preventing any servo output?

For flashing I don't have any jumpers on the BOOT pins at all, I've cut them off because they're not needed for flashing using STLINK. I don't have any other suggestions for you other than to try pulling master and flashing again. If it doesn't work then I dunno what else to tell you.

JupaCreations, 2021-10-01:
Hello Brian, I downloaed the latest version again.
Now it is working fine.

Thanks, Paco

closed 2021-10-01 by JupaCreations

CapnBry, 2021-10-01:
Awesome! Sorry about that, I completely released with a bug that made the one thing this was supposed to do not work. Glad updating resolved it for you.

JupaCreations, 2021-10-02:
Your work is appreciated even if it not works :-). It is the effort to find solutions to make other people happy.