CapnBry/CRServoF #15: Any reason this will not work with raspberry pi pico

mjsharma, 2021-10-08:
I am just asking because I have a few picos with me. Just want to make sure that Pico is powerful enough to do this.

CapnBry, 2021-10-08:
I do not know I have never used a Pi Pico, but I would assume it would if the code updated to support the Pico's timer mechanism. I assume their UART can do 420000 baud, yes?

James-Denton, 2022-04-03:
I've tried running this library on the RP2040 (both the Pi Pico and Arduino Nano Connect) using both the Mbed core and the earlephilhower core and each time it compiles with no errors, but when run the boards go into an error state (4 short then four long flashes of the inbuilt led)

Any ideas what might be causing it/how to fix it?

CapnBry, 2022-04-04:
> I've tried running this library on the RP2040 (both the Pi Pico and Arduino Nano Connect) using both the Mbed core and the earlephilhower core and each time it compiles with no errors, but when run the boards go into an error state (4 short then four long flashes of the inbuilt led)

There's no code to make it work with the RP2040? There's no environment for it, there's no target definition for it, and more importantly, this code doesn't do any blink patterns at all, so I have nothing to base my advice on to you.

That said, this is probably possible since the RP2040 appears to be able to actually do 420000 baud since it works with CrsfJoystick without any extra work.