CapnBry/CRServoF #16: Can this code working on arduino?

kkbin505, 2021-11-21:
If arduino nano or micro can do this?

CapnBry, 2021-11-21:
If you write code to make it work on an atmega it could, but out of the box it only supports STM32F103 processors. "Arduino" boards also can't run their UART fast enough for standard CRSF RX speed (420000 baud) so the receiver would have to be built with the new `RCVR_UART_BAUD` option in ELRS 2.0 to slow it down.

If you're asking if I'm going to make it work with Arduino atmegas, the answer is no.

kkbin505, 2021-11-22:
Thanks for your explanation. 
I have many Arduino boards, and just think if I can use arduino to do the same thing, becaust these boards are cheaper and smaller.
But if the atmega can not run UART fast enough, It seems difficult to do.
And Matek have release their CRSF to PWM board, it is not expensive.

CapnBry, 2021-11-22:
Matek also has an actual PWM receiver now too, the R24-P and there's a DIY 5ch version as well:
https://github.com/crteensy/ELRS-8285-1280-5xPWM

CapnBry, 2022-01-26:
The CrsfSerial is a library that can be used with the Arduino platform. Note that 420000 baud of standard receivers works with almost no Arduino hardware, so the receiver baud rate must be changed when flashing it.