CapnBry/CRServoF #18: i want use for teensy 4.0

H4S1M, 2021-12-09:
what do i need to change so i can use it on board teensy 4.0

CapnBry, 2021-12-14:
You'd need to write support for Teensy into it and add the target type for it. It isn't trivial because the servo class for that platform would also need to be validated to verify it doesn't have jitter in the output (the STM32 version did, which is why it is not used).

sfjuocekr, 2022-01-11:
I just finished porting it over to a Teensy 3.2, but I butchered out the servo stuff as I just needed the CrsfSerial class.

The code is very easy to port over, you just need to reimplement servoSetUs to fit your purpose.

Once I finish off the code for the joystick, would you be ok with me reusing your code @CapnBry?

H4S1M, 2022-01-12:
@CapnBry  i just want to get the pwm value for my teeensy fligh controller. How can i do it?

CapnBry, 2022-01-12:
> @CapnBry i just want to get the pwm value for my teeensy fligh controller. How can i do it?

https://github.com/CapnBry/CRServoF/blob/master/lib/CrsfSerial/examples/basic_crsf/basic_crsf.ino

@sfjuocekr It would be nice if you could clean it up and contribute it back but you can do whatever you want with the code as long as you publish it. It is GPL.

sfjuocekr, 2022-01-13:
I've just uploaded it, you had already made the changes that I did as well so I just pulled yours and called it a day.

https://github.com/sfjuocekr/CRSF-joystick/

This works using a Teensy 3.x board with any serial port, I use the second one.

wingunder, 2022-01-14:
Hi @sfjuocekr,

Sorry for butting in. I looked at your repository at https://github.com/sfjuocekr/CRSF-joystick
As you mentioned (only in this PR #18), it is a copy of https://github.com/CapnBry/CRServoF.
However, you forgot to do the following in your repository:

   1. Keep the LICENSE file.
       From GPL3:
 >You may make, run and propagate covered works that you do not
convey, without conditions so long as your license otherwise remains
in force. 

>You may convey verbatim copies of the Program's source code as you
receive it, in any medium, provided that you conspicuously and
appropriately publish on each copy an appropriate copyright notice;
keep intact all notices stating that this License and any
non-permissive terms added in accord with section 7 apply to the code;
keep intact all notices of the absence of any warranty; and give all
recipients a copy of this License along with the Program.

   2. Add proper recognition, _in your repo_, to @CapnBry for his work on https://github.com/CapnBry/CRServoF.
 
https://github.com/CapnBry/CRServoF is GPL3, and you _must_ follow the rules of GPL3, when using/copying it.
Please read, at least Chapter 5, in GPL3 carefully: Conveying Modified Source Versions.

Lastly, the preferred (and very simple) way to do this, is to simply fork https://github.com/CapnBry/CRServoF and then make your changes on the forked repo. Like this others can profit more from your work. They can for instance back-port your work into the origin, https://github.com/CapnBry/CRServoF, much easier. This way we can help each other and get to better, common code bases much quicker and easier.

Regards

sfjuocekr, 2022-01-14:
I forgot to add the license, credit was given in the main.cpp file.

If I might add, CrsfSerial and crc8 should then be moved to their own repository so you can include them as a library.

closed 2022-01-26 by CapnBry