CapnBry/CRServoF #20: My proto not running

pierrotm777, 2022-01-27:
Hello, thanks for this wonderful share !
I use this library https://github.com/stm32duino/BoardManagerFiles/raw/main/package_stmicroelectronics_index.json on a BluePill board.
I compile the sources under the Arduino IDE.
May be the libraries are bad for your system ?
For my test, i use a TBS TX/RX system.
So, my wiring is like that:
    TARGET_BLUEPILL pinout
    #define APIN_VBAT       A0
    #define USART_INPUT     USART2  // UART2 RX=PA3 TX=PA2
    #define OUTPUT_PIN_MAP  PA_15, PB_3, PB_10, PB_11, PA_6, PA_7, PB_0, PB_1 // TIM2 CH1-4, TIM3CH1-4

When i shutdown my transmitter, the green led is off and when my transmitter is on , the green led is on.
But, i have no PWM signal on the channel 1.
We try to add to our OpenAVRc transmitter the CRSF protocol and we are not certain of our code.
Thanks for your help,

Pierre

CapnBry, 2022-01-28:
I would think it would just work, since CRSF_CHANNELS_PACKED should be the same no matter what system is sending it. The LED on the bluepill should also only come on if we process the channels properly and therefore you should see PWM coming out of PA_15. Only thing I can tell you is to try putting in some debug logging or something?

pierrotm777, 2022-01-28:
Thanks you.
I try with Platformio and see what's happen

pierrotm777, 2022-02-07:
Hello, my proto is working now. Bad soldering was the reason. I can compile and upload with arduino ide and platformio.
Thanks for this nice share.

closed 2022-02-07 by CapnBry