CapnBry/CRServoF #22: SBUS output

sk8board, 2022-02-19:
Many fixed wing flight controllers require SBUS as an input. Can you add a change to output SBUS?

Thank you for your work on an ELRS converter.

CapnBry, 2022-02-19:
Why would this output SBUS? This is for outputting PWM, not digital.

closed 2022-02-19 by CapnBry

sk8board, 2022-02-19:
Because my fixed wing flight controllers require SBUS input, I have to use an ELRS receiver connected to a Matek converter (CRSF to PWM) then use another converter (PWM to SBUS) so I can use ERLS with fixed wing flight controllers.

If your software would convert from CRSF to SBUS, then I could eliminate the PWM to SBUS converter board. Why not go one step further to include SBUS on another pin?

CapnBry, 2022-02-19:
Because SBUS requires inverters and the STM32F103 does not have built in inverters, requiring an external inverter. Matek makes CRSF to SBUS boards now too (or they will very soon as I have seen them in the wild) if you need SBUS support.

sk8board, 2022-02-19:
Thank you for your help. I will wait for the Matek converter.