CapnBry/CRServoF #24: Failsafe

dxxiaoyun, 2022-10-31:
失控故障时，CH5和CH6没输出自定义PWM值，其它通道输出了自定的WPM值，是什么原因？

In case of out-of-control fault, CH5 and CH6 do not output customized PWM values, while other channels output customized WPM values. What is the reason?

CapnBry, 2022-11-01:
You must set the failsafe positions in the source code, did you do that?
https://github.com/CapnBry/CRServoF/blob/master/src/main.cpp#L14

Replace `fsaHold, fsaHold, fsaHold, fsaNoPulses` with the us failsafe values (e.g. 1500). The code is an example so it is set to show what options are available. If you want them to go to fixed values, just set them and re-upload the firmware.

dxxiaoyun, 2022-11-02:
收到您的回复我非常高兴！我测试了三次；
第一次测试“constexpr int OUTPUT_FAILSAFE[NUM_OUTPUTS] = {1500, 1500, 988, 1500,  998, 998, fsaHold, fsaNoPulses };”
故障时CH5，CH6 输出为保持当前状态，没有输出PWM(998),CH1~4)输出了我设定的PWM值。
第二次测试 ”constexpr int OUTPUT_FAILSAFE[NUM_OUTPUTS] = {1500, 1500, 1400, 1500, fsaHold, fsaHold,998, 998 };“
故障时CH1~4,CH7~8)输出了我设定的PWM值，所以我认为故障时CH5，CH6不能执行自定义值 ，向您求功。
第三次测试 “constexpr int OUTPUT_FAILSAFE[NUM_OUTPUTS] = {1500, 1500, 988, 1500, 998, 998,998, 998 };”
故障时所有通道都输出了自定义的PWM值，功能完全正常测试成功，非常感谢您的创作！
I am very happy to receive your reply! I tested it three times; First test "constexpr int OUTPUT _ FAILSAFE [NUM _ OUTPUTS] = { 1500, 1500, 988, 1500, 998, 998, fsaHold, fsaNoPulses };" In case of fault, the output of CH5 and CH6 is to maintain the current state, and no PWM (998) is output, and the PWM value set by me is output by CH1 ~ 4). The second test "constexpr int OUTPUT _ FAILSAFE [NUM _ OUTPUTS] = { 1500, 1500, 1400, 1500, fsaHold, fsaHold, 998, 998 };" In case of failure, CH1 ~ 4, CH7 ~ 8) output the PWM value I set, so I think that CH5 and CH6 cannot execute the custom value in case of failure, and I ask you for work. The third test "constexpr int OUTPUT _ FAILSAFE [NUM _ OUTPUTS] = { 1500, 1500, 988, 1500, 998, 998, 998 };" In case of failure, all channels output the custom PWM value. The function is completely normal and the test is successful. Thank you very much for your creation.

CapnBry, 2022-11-02:
It doesn't make any sense though, the code treats all the servos the same so there's no reason if CH5 and CH6 operate normally, then they should go to failsafe as the rest do. I don't have a receiver hooked to my CRServoF any more so it seems like a lot of work for me to confirm this. I've checked the code and it is so simple I can't imagine it is going wrong.

closed 2023-12-18 by CapnBry