CapnBry/CRServoF #25: Adapting to Pro Micro without "#include <functional>"

robhaswell, 2022-11-11:
I'm trying to adapt CRSFJoystick, and so also this library, to run on a SparkFun Pro Micro. The error I am running into is:

```
CrsfSerial/CrsfSerial.h:4:10: fatal error: functional: No such file or directory
```

This means that I can't use the `std::function` syntax to declare the event handlers.

My software experience is primarily Python so I am struggling with the deep magic here, would you be able to help me get this compiled for my platform please?

CapnBry, 2022-11-14:
I removed `<functional>` back in May, man. Catch up! :-D

robhaswell, 2022-11-14:
Lol I literally just now discovered this and was about to close this issue! Thanks for getting back to me.