CapnBry/CRServoF #36: CrsfSerialStream undefined

sk8board, 2023-12-19:
Hello Bryan,

I am attempting to compile the CRServoF project but I keep getting an error that states CrsfSerialStream is undefined. I have looked through the following files and have determined the error is accurate. 

CrsfSerial.h, CrsfSerial.cpp, and Crsf-protocol.h

[image]

Can you tell me what is CrsfSerialStream and where is it defined?

Thank you,

sk8board, 2023-12-19:
I know you are going to state that this project is not finished for Pico, but I have a project for an RC crawler receiver which will support LED lighting that can be programed using EdgeTX.

The correct definition for Pico UART is as follows.

[image]

I know this would need to be further developed with a scheme that will work for the other microprocessors that were originally included in this project. TBD.

It looks like the Pico servo code was for Mbed but needs to be changed to arduino-pico (earlephilhower). TBD.

I will let you know when it is done.

CapnBry, 2023-12-19:
The UART isn't the problem, it is the Servo library. I didn't look very hard but the only one I saw was the great one by `earlephilhower`. I don't want to use that though because it doesn't compile on windows without jumping through hoops to increase the path length and that just creates more headaches for me as I get support question after support question wondering why it won't build.

If you manage to make servo output without using that platform, then let me know!

sk8board, 2023-12-19:
To avoid servo jitter issues while using a Pico, the PIO function of the Pico needs to be used, as mentioned here. 

https://github.com/earlephilhower/arduino-pico/issues/1396

If you are having problems with compiling the `earlephilhower` code, you should submit a bug so it can be addressed. This servo code is a good base to start with. It does not make sense to reinvent this work.

Also, the reason that I submit here is because you have disabled the discussion portion of this Github.

CapnBry, 2023-12-20:
Oh, I'm aware of how to use it, but that core requires a user has to get into their windows PC's system settings and enable some stuff. I don't want to use it because of that. If you don't follow their instructions, it doesn't build. It isn't a bug, it is their requirement. 

I don't have a discussion portion because I don't want to have a discussion about anything. This project was a proof of concept so it should not be considered active. The only thing I care to keep working on is the CRSFSerial library I wrote for it, which does work on the PiPico by itself.

sk8board, 2023-12-21:
FYI: If you plan to complete development for Pico, the `earlephilhower` base code is not compatible with the base code that you used for the other micro's. This would require many if(STM32)-else(Pico) statements to add to your code to get it to work. This was getting too messy for my preference. 

I found the [mikeneiderhauser/CRSFJoystick](https://github.com/mikeneiderhauser/CRSFJoystick) project was easy to use. The Gamepad code did not work with the `earlephilhower` base code. Therefore, I removed the Gamepad code and added the `earlephilhower` servo libray. I had to add a little code for the servos and RC crawler LED's to the main and include files.

If someone is bugging you about using your code with an PR2040-Pico, then I recommend sending them to `mikeneiderhauser/CRSFJoystick`.

CapnBry, 2023-12-21:
haha but that's just **this** code with all the servo stuff removed and gamepad added! I think you'd have ended up the same place if you just set this project to use the standard pio, then added the Servo lib too
```ini
[env:pico]
platform = raspberrypi
board = pico
framework = arduino
```

What servo library did you use? I got derailed way back when I was messing with supporting it and all I found at the time was earlephilhower's.

sk8board, 2023-12-21:
Yes, you are correct, but...

```
[env:pico]
platform = raspberrypi
board = pico
framework = arduino
```

This `platformio.ini` setting uses `mbed` base code for the Pico. The problem is I want to take advantage of the PIO PWM feature of RP2040, but `mbed` does not support PIO. 

https://github.com/earlephilhower/arduino-pico/discussions/246#discussioncomment-990782

There are only two servo libraries I found that use PIO.

One servo PIO project is `metanav` [https://github.com/metanav/pico_servo_pio](url), which uses `Pico-SDK`. 

The other is `earlephilhower`, which requires:

```
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower
```
When changing to this `platformio.ini` setting in your code, I run into many compile errors.

When I make this `platformio.ini` change using [mikeneiderhauser/CRSFJoystick](https://github.com/mikeneiderhauser/CRSFJoystick), the only problem that I have is with the Gamepad library.

If you want to use the PIO PWM function of RP2040, then the easy way is to use the servo library from `earlephilhower` with [mikeneiderhauser/CRSFJoystick](https://github.com/mikeneiderhauser/CRSFJoystick).

I am sure you can make the Pico work with your code. I believe the STM32 code does not play nice with `earlephilhower` which was specifically developed for the RP2040 Pico. Since STM32 and RP2040 are quite different, it seemed easier and cleaner to have separate projects for these two micro's.

CapnBry, 2023-12-21:
Thanks for the extra info, when I looked at adding it, I tried to just copy paste the Servo files over and yolo but it complained about the coprocessors program definition so I didn't go any further. I'm about to head out for the holidays but if you got it to compile then obviously this project can compile because CRSFJoystick is mostly just this project.

I may have some questions about your implementation for you next week!

sk8board, 2023-12-22:
I have posted my files for you to use if needed.

https://github.com/sk8board/CRServoF_for_RP2040_Pico

The differences from your project are as follows:

`main.cpp`, `include`, and `platformio.ini` files are from [mikeneiderhauser/CRSFJoystick](https://github.com/mikeneiderhauser/CRSFJoystick) which were modified for my RC crawler application.

The servo library is from `earlephilhower`, https://github.com/earlephilhower/arduino-pico/tree/master/libraries/Servo/src

I think you will find the `earlephilhower` code base requires a significant change to your `main.cpp`, which I thought was easier to manage with a separate projects, STM32 vs RP2040.

Thank you for your contribution to the ELRS project. ELRS is awesome!

CapnBry, 2023-12-22:
Thanks! I've looked at your repo and looks like you're using _the entirety of_ `earlephilhower`, not just the Servo library?
```ini
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board_build.core = earlephilhower
```

That's what I want to avoid using, because of all the extra installation steps and global settings changes for windows. I can't get the Servo library to compile on its own due to all the dependencies in the custom platform.

sk8board, 2023-12-23:
If you are concerned about people contacting you about long path, you could add a link to the Windows configuration information like I did on my GitHub.

https://arduino-pico.readthedocs.io/en/latest/platformio.html#important-steps-for-windows-users-before-installing

With regard to using a less efficient servo library, you could offload the servo functions to the second core of the RP2040. Unless calling the second core, all code on the RP2040 is executed by the first core.

https://datasheets.raspberrypi.com/pico/raspberry-pi-pico-c-sdk.pdf#%5B%7B%22num%22%3A30%2C%22gen%22%3A0%7D%2C%7B%22name%22%3A%22XYZ%22%7D%2C115%2C461.71232%2Cnull%5D

https://www.raspberrypi.com/documentation/pico-sdk/high_level.html#pico_multicore

closed 2023-12-26 by CapnBry

sk8board, 2024-01-21:
I ran into a limitation with the `earlephilhower` servo library, so I went looking for an alternative, again. I discovered the Pico C-SDK has PWM functions already built into the base code which use the PIO state machines. But, the documentation is lacking... 

I eventually discovered how to use these PWM functions and documented this work [here](https://github.com/sk8board/Duty_Cycle_and_Servo_PIO_for_RP2040_Pico). Now you can use your CRSF library on a Pico without `earlephilhower` base code and their long paths.