CapnBry/CRServoF #37: Adding ESP32 S2 support target hw: lolin_S2_mini

helicocrasher, 2023-12-29:
@CapnBry  
Thanks for that code.

I managed to add get the CRSF link running on the wwemos LolinS2_mini.
The only challenge was to feed the used serial port in numbers to CRSR.begin() function, as on the ESP32 S2 they are not fixed.
Seen a lot of beauty in the code ... CRSF serial telemetry, I learned a lot. 
Now merging some ESP HW PWM code to it.

helicocrasher, 2024-01-02:
ESP32 HW PWM now also working. 8 Channels on ESP32 S2

robhaswell, 2024-03-20:
> ESP32 HW PWM now also working. 8 Channels on ESP32 S2

Where is the code for this, perhaps you could create a PR? I'm interested in running this on an ESP32-C3.

robhaswell, 2024-03-20:
Thanks, I'll take a look. My ulterior motive is to submit a PR to this project which adds VTX admin, VTX arming, and potentially a very limited implementation of voltage OSD. The goal being to be able to use this in a fixed wing without buying a full flight controller. How would you feel about those PRs?

CapnBry, 2024-03-20:
> How would you feel about those PRs?

I don't really want them 😅 I don't want to maintain this project considering ExpressLRS supports PWM output itself now so this project is obsolete.

robhaswell, 2024-03-20:
> > How would you feel about those PRs?
> 
> I don't really want them 😅 I don't want to maintain this project considering ExpressLRS supports PWM output itself now so this project is obsolete.

I get that lol, problem is that ExpressLRS doesn't want to add VTX admin, but also a user might not want to buy a PWM-capable ELRS receiver and might find adding a $2 ESP devkit to be more attractive. The main motivation is not being able to get VTX admin in ELRS though.

I get it though, I'll create a fork if I ever get round to this.

CapnBry, 2024-03-20:
If you wait long enough I bet ELRS will add smartaudio output. The receiver code is becoming more and more full of supporting everything a flight controller does so I think it is just a matter of time that everyone agrees SA support is worth it.

robhaswell, 2024-03-20:
Good to know! Thanks!