CapnBry/CRServoF #38: Failsafe Adjustment

sk8board, 2023-12-30:
Hello Bryan,

Is there an easy way for the user to adjust failsafe settings?

For example, could the ELRS team "simply" add the [failsafe WebUI](https://www.expresslrs.org/hardware/pwm-receivers/?h=failsafe#channel-mapping-and-failsafe) configurator to select non-PWM receivers?

Or, if that is not so simple, then could a LUA script be used to adjust Failsafe settings over UART from the receiver to the BluePill or RP2040?

CapnBry, 2024-01-03:
No the failsafe is completely external and compiled in so there's no way to adjust it after the fact.

Adding failsafe to the ExpressLRS code will be done at some point, but I don't know how that will look because standard receivers will only have like 3 lines on the webui. I really don't want to think about it at this time.

I can tell you we will NEVER support publishing failsafe values to an external device as a configuration thing so I don't think you'll ever get what you want. You could implement CRSF config protocol in your project, then add failsafe values as config items and then it will just work with the existing Lua config system.

If your project is a one off then it isn't too difficult to just compile in the failsafe or come up with your own way to configure it. If you're trying to make a product using this code, I would highly discourage that and you'd be entirely on your own. 

I will tell you again. This project is not in active development. I have no desire to start a new project. This project was created simply to show how it can be done because there were zero solutions available at the time. There are many solutions now so this project is just around for posterity and the CrsfSerial library.

sk8board, 2024-01-03:
> You could implement CRSF config protocol in your project, then add failsafe values as config items and then it will just work with the existing Lua config system.

I appreciate your input. Is there any documentation of how to utilize the existing Lua config system or CRSF config protocol?

CapnBry, 2024-01-03:
https://github.com/crsf-wg/crsf/wiki/Config-Protocol