CapnBry/CRServoF #39: Use CrsfSerial with Arduino RP2040 Connect

kingfisher09, 2024-01-20:
Hi,

I'm building a robot using an Arduino RP2040 Connect. I want to control the robot using ELRS. I found this project which uses an Pi Pico [CRSFJoystick](https://github.com/mikeneiderhauser/CRSFJoystick) that uses your library so it seems like your library should work with the RP2040 chip. I've uploaded the basic_crsf example to the board, it compiles just fine but I'm not seeing a signal. I'm not sure whether I'm using the right RX and TX pins on the board. 

I see in the code you can select Serial1 or Serial2 but I'm not sure what they refer to. The [Arduino pinout ](https://docs.arduino.cc/static/a4d65079ebe3058e37153eb0f93890a8/2f891/pinout.png)only lists one RX and TX so I tried that first but I also see that on the [Pi Pico pinout](https://cdn-learn.adafruit.com/assets/assets/000/099/339/original/raspberry_pi_Pico-R3-Pinout-narrow.png?1612915004) there are other pins for other serial ports so I tried connecting to the equivalents of those pins. 

Is there something else that I'm missing?

Thanks

CapnBry, 2024-01-20:
I don't have Pi Pico RP2040 Servo output support. The CrsfParser library does support it, but there is no Servo output. See the other Issues about it and there's a link to someone who copy pasted a bunch of code to make it work.

closed 2024-01-20 by CapnBry

sk8board, 2024-01-20:
I believe this is what you are looking for with regard to serial.

https://arduino-pico.readthedocs.io/en/latest/serial.html

Also, here is the compilation that Bryan was referring to.

https://github.com/sk8board/CRServoF_ELRS_for_RP2040_Pico

Good luck fishing ;)

Shabbat Shalom

kingfisher09, 2024-01-20:
Hi, thanks so much for getting back to me. I tried to create a new board entry in the board_defs.h file but when I upload the code to the board it hangs. 

I tried stripping down the code to the absolute bare minimum:
I tried to stop the code down to the absolute bare minimum which gave me this:
```
#include <CrsfSerial.h>
#define UART_SELECT Serial2
#define CRSF_TX 0  // physical 6 (UART1 == Serial2)
#define CRSF_RX 1  // physical 7 (UART1 == Serial2)
CrsfSerial crsf(UART_SELECT, CRSF_BAUDRATE);  // pass any HardwareSerial port

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  // UART_SELECT.setTX(CRSF_TX);
  // UART_SELECT.setRX(CRSF_RX);
}

void loop() {
  // put your main code here, to run repeatedly:
  Serial.print("Channel 1  ");
  Serial.print(crsf.getChannel(1));
  Serial.println();
  delay(100);
}
```
Annoyingly this still locks up the board. I experimented by commenting out different lines to see what was causing the freeze. I eventually found that it seems to be the
`UART_SELECT.setTX(CRSF_TX);`
`UART_SELECT.setRX(CRSF_RX);`
lines that are causing the problem. Have I misunderstood how to use this code?

Thanks again!

kingfisher09, 2024-01-20:
Ok, I have absolutely no idea what I did differently but I just tried the basic_crsf sketch again and it worked! Yay!