CapnBry/CRServoF #43: Add RPM, TEMP sensors

pierrotm777, 2025-05-20:
Hello,

I see on this [ExpressLRS](https://github.com/ExpressLRS/ExpressLRS/pull/3197) link that you have helped to add RPM and TEMP options.
Can you add these options also into your code here ?

Thanks,

Pierre

CapnBry, 2025-05-20:
Where would these values come from? There is no external telemetry supported and CRServoF doesn't have any temperature or RPM sensors itself.

pierrotm777, 2025-05-20:
You have developed a lib named CrsfSerial that I use for send voltage, current, gps, vario to my handset. I goal would be to update CrsfSerial with temp, rpm.

CapnBry, 2025-05-21:
https://github.com/CapnBry/CRServoF/blob/main/lib/CrsfSerial/examples/telemetry/telemetry.cpp

Note this isn't in EdgeTX yet, so there's nowhere for the values to go without writing your own lua to parse them back out. Example of that is in the PR you linked.

pierrotm777, 2025-05-21:
Really thank you for your job.
I love this library that I use from two years ago.
Have a good day,
Pierre

closed 2025-05-24 by CapnBry